jneilliii/OctoPrint-ConsolidatedTabs #4: Webcam Tab does not work

bcorcoran, 2020-08-06:
Using https://github.com/malnvenshorn/OctoPrint-WebcamTab and if I place it inside of the Consolidated Tab, the webcam image never loads. I think the javascript required for it to function is not getting run.

jneilliii, 2020-08-06:
I'll have to take a look at that one. I've included all the initialization callbacks that normally happen on tab change and after tab change, so not sure what else I might need to add for that plugin to work properly with mine.

jneilliii, 2020-08-06:
I guess my question would be do you need that plugin is you move the control tab to the consolidated on anyway?

bcorcoran, 2020-08-06:
Yeah, I prefer it being in its own tab so that I can't accidentally use the controls.

jneilliii, 2020-08-06:
Fair point. I disable the option `Enable Keyboard Control` in OctoPrint's settings. I'll still take a look when I get some time.