jneilliii/OctoPrint-ConsolidatedTabs #6: Temperature dropdown only shows one value

beecreate, 2020-08-07:
Both the Bed and the Hotend temperature dropdowns seem to drop behind a whitespace.
I have 3 values, theres only room to show the first

see: 

[image]

jneilliii, 2020-08-07:
Thanks, I'll look into this. Probably just needs a little CSS magic to allow overflow.

rorosaurus, 2020-08-08:
I can confirm this issue, and it appears even if the temperature window is tall enough.

jneilliii, 2020-08-13:
I've made some updates to the plugin in a separate branch, that should resolve this. If you want to try it out uninstall the current version and install using the URL below in plugin manager > Get More > ...from URL. I also enabled a selecting option that allows you to select a panel to bring it to the front. Only the selected tab is resizable. 

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

Also, just as general feedback, what are your thoughts on removing the draggable capability all together and just make the panels sizable? All the panels would then just flow in the order they are in the plugin's settings, wrapping once they go beyond the tab container.

jneilliii, 2020-08-13:
Oh, this new version changes the logic for the width too, and there's a new setting for either leaving it default width or making it full width of the browser.

beecreate, 2020-08-13:
Thanks for your quick reply.

At first i thought it didnt work, however after re-reading your instructions, I figured out i had to select the tab first and THEN i can see the resize. (You can click the cross and drag the tab without first selecting it).
It works great, however, i would suggest adding those instructions in the plugin tab.

The 'use browser width' is perfect. that was the only thing i used Themefy for, so i can remove that pluging.

The general feedback question; sounds like a nice idea, but personally i dont have a need for it. (but i'm more of telling the computer what i want, not letting the computer decide what it thinks i want', especially with layout ;) )

It would be great if you could add a 'freeze' that would keep the current setting (scaling/ position / z-order) unless you 'unfreeze'.
Because currently it can happen that you drag something unintended. (but you're free to call this issue closed and just let me create a new feature request for it)

jneilliii, 2020-08-13:
Yes, the freezing option is something that I intend to add for sure. Thanks for the feedback. I did add the note to the readme, and I will probably add a link to those notes in the plugin's settings as well prior to official release of this version.

Jarren141, 2020-08-14:
Can confirm that this update fixed the issue. Thanks for sharing the great work!!!

rorosaurus, 2020-08-14:
I love it! I have some feedback:

1. I think beecreate mentioned this already, but just to recap: If you directly click on the temp dropdown, while some other tab (like Terminal) was selected, this bug reproduces. You must first select the "Temperature" tab, then the dropdown, in order to see the dropdown.

2. If GCode Viewer is in your consolidated tab, and you selected to use the full width, there's a small useless horizontal scroll bar in your browser. (Or mine, at least!)

3. When dragging the tab around, and often when letting go, my browser becomes "busy" (cursor changes to hourglass and locks up for a few seconds). I don't remember this in the previous version.

4. You can't reposition the bottommost tab further downward. I personally wanted to add just a bit of padding between each tab, but it breaks when I get to the last one. :)

5. I'd be fine removing the dragging capability. I think I'd prefer it uses the order from settings, like you suggested, but I understand other users might want to customize it more. But there's definitely something to be said about trying to simplify the scope of the plugin, before it becomes a full-blown window manager! 😅

jneilliii, 2020-08-14:
> 1. I think beecreate mentioned this already, but just to recap: If you directly click on the temp dropdown, while some other tab (like Terminal) was selected, this bug reproduces. You must first select the "Temperature" tab, then the dropdown, in order to see the dropdown.

This was the best I could do for now. I think I can adjust the selectable option I pushed an update for last night to maybe auto select the panel when you click the drop down. 

> 2. If GCode Viewer is in your consolidated tab, and you selected to use the full width, there's a small useless horizontal scroll bar in your browser. (Or mine, at least!)

That's part of the viewer I think, the scroll bar is the layer height indicator. 

> 3. When dragging the tab around, and often when letting go, my browser becomes "busy" (cursor changes to hourglass and locks up for a few seconds). I don't remember this in the previous version.

It wasn't in the previous version. It was added to indicate that a change is saving after dragging/ resizing. 

> 4. You can't reposition the bottommost tab further downward. I personally wanted to add just a bit of padding between each tab, but it breaks when I get to the last one. :)

Looks like I inadvertently removed the resizer in the tab container last night. I'll put that back in.

> 5. I'd be fine removing the dragging capability. I think I'd prefer it uses the order from settings, like you suggested, but I understand other users might want to customize it more. But there's [...]

Still on the fence about this.

jneilliii, 2020-08-14:
I just pushed updates to the 0.0.8 branch that requires activating a panel using shift+click for enabling dragging/resizing. This update also allows for clicking on panels to bring them to the front, which will allow for seeing temperature menus if they extend beyond the boundary of the panel. This is just a minor change to the version you installed @rorosaurus I believe, switching from holding the shift key down instead of the ctrl key.  This resolved some weirdness with certain events where a previously selected tab couldn't be activated.

I looked into adding an option to the selectable to select the temperature panel when a button was clicked (that's what the menu is) and unfortunately, it killed the menu altogether and it wouldn't open. I'll have to loo into other methods of handling that weirdness of having to select the panel first.

miobx, 2020-08-17:
The issue of the temp drop down staying behind a tab below the temp tab still exists for me.  using 0.0.8
[image]

jneilliii, 2020-08-17:
Click on a white area next to the drop down in that panel to being it to the front.

miobx, 2020-08-17:
I tried your suggestion and it still displays behind the panel below the temp tab

jneilliii, 2020-08-17:
Can you try this. Click 1 and then click 2.

[image]

miobx, 2020-08-17:
Still the same thing......
[image]

I am using Chrome Browser Version 84.0.4147.105 (Official Build) (64-bit) if that helps.

miobx, 2020-08-17:
OK wait........If I open the drop down then click on white space after the drop down is open it comes to the front.    If that was the direction above I missed the sequence......

jneilliii, 2020-08-17:
Ok, that was not what I was expecting, but I can see now why that might be the case. There's something about the click events that get blocked on certain items and not on others as part of the jQuery selectable library. That must be at play here. Glad you got a temporary workaround until I can figure out how to select that panel when the menu opens up.

jneilliii, 2021-03-06:
So with recent development of my gridstack implementation the drop downs cause the panel to become scrollable, which allows you to get to all the items. I feel that's a fair compromise. If you want to test it out you can copy/paste the URL below into Plugin Manager > Get More > ...from URL and clicking install. Once installed you'll need to re-add your tabs and arrange the way you like.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

closed 2021-03-07 by jneilliii