jneilliii/OctoPrint-ConsolidatedTabs #7: Compatibility with Themeify plugin

pszeder, 2020-08-07:
I use the Themeify plugin to change the make the UI more eye friendly. 
The CSS of this plugin causes the backgrounds of the consolidated tabs to be white as seen below:
[image]

For the time being I've removed the line for the background of the .panel and .panel-heading in CSS to have this: 
[image]

I think this could be a good and quick fix which would work with the other themes too, but of course, would change the look without Themeify. Another possibility would be to fix it in Themeify, but I'm not sure what their opinion is, other than try to not set colours in other plugins.

jneilliii, 2020-08-07:
You just need to add your own advanced options...

[image]

Relevant css selectors...

[image]

jneilliii, 2020-08-07:
Added the above tips to the readme.

closed 2020-08-07 by jneilliii

pszeder, 2020-08-07:
I came across advanced options of Themeify after submitting this issue, but didn't think it would be the preferred way as it's not something an average user can do. Anyway having it in the readme will at least give them a chance to set it up correctly.