jneilliii/OctoPrint-ConsolidatedTabs #8: Mac OSX Catalina Chrome and Safari Problems

murathanaraz, 2020-08-07:
Hello, very good idea however too buggy on Safari and Chrome on Mac, it doesn't work properly. Kind regards,

jneilliii, 2020-08-07:
Well, if you want it supported then I'll need more information than just "it's too buggy". Otherwise, you could buy me a mac so I can develop on that.

murathanaraz, 2020-08-07:
Sorry if sounded offensive, it was not my intention. On safari the left side contains the connection and files information is there but the tabs container is one break under the that side container. When you drag tabs, the other tab goes off the container and when you click that it goes back. On safari it is totally unusuable. 

On chrome both containers are in their places however the tabs are inconsistent. As safari is my main browser I needed to disable it until the next update comes.

And If you are interested in business please let me know your email address. I have a plug-in idea to be developed.

Cheers

jneilliii, 2020-08-07:
Thanks, that sounds similar to #3. One question, have you set your overall width value in the plugin's settings? Some of the users on that other issue reported that can possibly help your situation.

murathanaraz, 2020-08-07:
I know CSS and HTML and I tried all options available, didn’t help.

jneilliii, 2020-08-12:
Yeah, this is a tricky one. Can you reproduce the same issue by uninstalling the plugin and installing my dev version using the link below in Plugin Manager > Get More > ...fromURL and clicking Install? After install go into settings and check the box for full width.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

murathanaraz, 2020-08-12:
Attachment available until 11 Sep 2020
I have captured the screen for you. And sending the video with mail drop. Let me know if you can’t get the video
Click to Download
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAZpn3V_hdbF6FFPgaq6jwIGE6ptvAVeEJltzfz3a083Un2Nz1ZX3ZpIu%2F%24%7Bf%7D%3Fo%3DAp3vWLKjj7pp1CgEl8nl8Z3cpFWk4FW1K13iXK_coWO7%26v%3D1%26x%3D3%26a%3DCAogHE0ir_X2X1PTbEsKDdRYkVc6Tzh4rpw64qnRBUvkBU0SdRC0jdadvi4YtJ3R8ccuIgEAKggByAD_Gib27lIEhOqbb1oE92aSLmolZEL3n7sLMHFvxn1vLlgpGj9dhxCqHHJlmWQ3tWoP91mLT8rtcnIlUw5glQdAbc9GSxHAavznOiyNlB_qsmtitVgayhpzOT2QchMxJw%26e%3D1599845191%26fl%3D%26r%3DC605C225-FCC3-4426-8029-62B1A8940F8D-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3D8AD521BC-2454-472A-B239-01AE885085D3%26p%3D45%26s%3Dd7c9012nmqBdvY_0_TV0SzFlk8k&uk=lETg1Vxr4_Hov_D1mXeOXg&f=Screen%20Recording%202020-08-12%20at%2018.22.21.mov&sz=67366914>Screen Recording 2020-08-12 at 18.22.21.mov
67.4 MB

> On 12 Aug 2020, at 15:10, jneilliii <notifications@github.com <mailto:notifications@github.com>> wrote:
> 
> 
> Yeah, this is a tricky one. Can you reproduce the same issue by uninstalling the plugin and installing my dev version using the link below in Plugin Manager > Get More > ...fromURL and clicking Install? After install go into settings and check the box for full width.
> 
> https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip <https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip>
>

murathanaraz, 2020-08-12:
Hi, I have captured the screen for you. Please feel free to ask questions, I am happy to help the development.

https://wetransfer.com/downloads/2b1ebd93a1b674e8dd6818366e1edc2720200812172719/43667408eeefdd6b841f5b523e6f4b9d20200812172744/94409b <https://wetransfer.com/downloads/2b1ebd93a1b674e8dd6818366e1edc2720200812172719/43667408eeefdd6b841f5b523e6f4b9d20200812172744/94409b>

jneilliii, 2020-08-12:
Yeah, the plugin doesn't like being resized currently at all. Thanks for the video that makes things much easier to understand. I'm still working on this plugin, so hopefully once I find a good CSS person I can get this fixed soon.

murathanaraz, 2020-08-12:
Sure, however, the octoprint containers do not like resize as well. Containers should be displayed as block so one can drag and change the order only. Their distance to each other should be fixed distance as well. These are just my thoughts though, 

If you need testing again let me know

jneilliii, 2020-08-13:
What are your thoughts on removing the draggable capability all together and just make the panels sizable? All the panels would then just flow in the order they are in the plugin's settings, wrapping once they go beyond the tab container.

murathanaraz, 2020-08-13:
That would be the best and hassle free solution I guess.

Depending on your browser width you can see 2 or 3 tabs in a row... to be honest mostly 2 is enough for me. Control and Temperature tabs are the ones I want to see on same page due to the camera view.

murathanaraz, 2020-08-18:
Attachment available until 17 Sep 2020
I have updated the plugin. This is how it looks on Safari Mac OSX Catalina. I don’t know if it is how you wanted to be but looks like they should be side by side on full size window. Fyi.

Click to Download
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAb4c0PjknP3QCR9Glg7DwM_HeY7oAcZnhj7-DIMPrMuDDuhtKn6asRBy%2F%24%7Bf%7D%3Fo%3DAsSehvooidpCT7BSZ-MlRZsExkMVE_v8pwRBBeGdeR-a%26v%3D1%26x%3D3%26a%3DCAog71CeY1j-OUn7GiZuvXlcF3FDWjDG5wTFEcs8y5Tq8DASdRDj3a-VwC4Y4-2q6ckuIgEAKggByAD_HjwTR1IEx3mO6FoEmrEQcmoltLEKZA9-laaMUDbeOrICRcU-f-y0SlpsnVePWeauQXB4mWdF7nIlvb8ldJhsq_Y4fFNGVafk9R_QPCEi_IE2ccNf6ZLBL7tKJDLq0A%26e%3D1600364656%26fl%3D%26r%3D5822E940-2944-434C-98F1-2E18D1D745FF-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3D8AD521BC-2454-472A-B239-01AE885085D3%26p%3D45%26s%3DlSZat4gZZMryX_GEOWtFYmtpSmU&uk=tEP1um_RpnYHpDEsWfkJ7g&f=Screen%20Recording%202020-08-18%20at%2018.40.17.mov&sz=19816132>Screen Recording 2020-08-18 at 18.40.17.mov
19.8 MB

jneilliii, 2020-08-18:
That was intentional. Documented in the release notes, and on the readme under tips. Hold down your shift key and click a panel and it will become movable and resizable.

murathanaraz, 2020-08-18:
Cool. Much better control now.

Thanks

jneilliii, 2020-08-19:
Just released 0.0.9 version that adds additional description in the settings on how to activate a panel for moving and resizing.

closed 2020-08-19 by jneilliii

murathanaraz, 2020-09-05:
Hello, can you contact me via private email? venedikli@gmail.com