jneilliii/OctoPrint-ConsolidatedTabs #9: tabs disappear

kb-, 2020-08-08:
After selecting the tabs to group and refreshing the page, the selected tabs just disappear. They don't show up grouped.
It might be an incompatibility with another plugin.

I'm on OctoPi Version 0.15.1, running on Raspberry Pi 3 Model B Plus Rev 1.3
With the following plugins installed:

  | OctoPrint: 1.4.0                                       Installed: 1.4.0                     Available: 1.4.2 (stable release)                     Release Notes: https://github.com/OctoPrint/OctoPrint/releases/tag/1.4.2
--
  | Active Filters: 0.0.1                                       Installed: 0.0.1                     Available: 0.0.1                     Release Notes: https://github.com/MoonshineSG/OctoPrint-ActiveFilters/releases/tag/0.0.1
  | Autoscroll: 0.0.3                                       Installed: 0.0.3                     Available: 0.0.3                     Release Notes: https://github.com/MoonshineSG/OctoPrint-Autoscroll/releases/tag/0.0.3
  | Consolidated Tabs: 0.0.7                                       Installed: 0.0.7                     Available: 0.0.7                     Release Notes: https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/releases/tag/0.0.7
  | Custom Control Editor Plugin: 0.2.2                                       Installed: 0.2.2                     Available: Version 0.2.2                     Release Notes: https://github.com/Salandora/octoprint-customControl/releases/tag/0.2.2
  | Emergency Stop Button: 0.1.2                                       Installed: 0.1.2                     Available: 0.1.2                     Release Notes: https://github.com/ntoff/OctoPrint-Estop/releases/tag/0.1.2
  | Fullscreen Plugin: 0.0.5                                       Installed: 0.0.5                     Available: 0.0.5                     Release Notes: https://github.com/BillyBlaze/OctoPrint-FullScreen/releases/tag/0.0.5
  | Print History Plugin: 1.2                                       Installed: 1.2                     Available: Slicer parser by @armicron                     Release Notes: https://github.com/imrahil/OctoPrint-PrintHistory/releases/tag/1.2
  | Print Time Genius Plugin: 2.2.5                                       Installed: 2.2.5                     Available: Add option to disable gold stars                     Release Notes: https://github.com/eyal0/OctoPrint-PrintTimeGenius/releases/tag/2.2.5
  | PrusaSlicer Thumbnails: 0.1.2                                       Installed: 0.1.2                     Available: 0.1.2                     Release Notes: https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/releases/tag/0.1.2
  | Python 3 Check: 0.1.4                                       Installed: 0.1.4                     Available: 0.1.4                     Release Notes: https://github.com/jneilliii/OctoPrint-Python3PluginCompatibilityCheck/releases/tag/0.1.4
  | Stateful Sidebar: 0.1.2                                       Installed: 0.1.2                     Available: 0.1.2                     Release Notes: https://github.com/jneilliii/OctoPrint-StatefulSidebar/releases/tag/0.1.2
  | System Command Editor Plugin: 0.3.4                                       Installed: 0.3.4                     Available: Version 0.3.4                     Release Notes: https://github.com/Salandora/OctoPrint-SystemCommandEditor/releases/tag/0.3.4
  | TerminalCommands Plugin: 0.1.6                                       Installed: 0.1.6                     Available: 0.1.6                     Release Notes: https://github.com/ieatacid/OctoPrint-TerminalCommands/releases/tag/0.1.6
  | Themeify: 1.2.2                                       Installed: 1.2.2                     Available:                      Release Notes: https://github.com/Birkbjo/OctoPrint-Themeify/releases/tag/v1.2.2

jneilliii, 2020-08-08:
Please try this version by using the URL in Plugin Manager > Get More > ...from URL and click Install. Once installed go into the plugin's settings and click the clear positions and clear sizes button and enable the new 'Use Full Width of Browser' option.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

kb-, 2020-08-08:
This version fails installing unfortunately. I get the following message "Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output"

[install.log](https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/files/5046099/install.log)
[plugin_pluginmanager_console.log.txt](https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/files/5046102/plugin_pluginmanager_console.log.txt)

jneilliii, 2020-08-08:
Did you copy/paste the URL into the ...from URL input? There are people that have successfully installed this working on the other issue.

kb-, 2020-08-08:
Yes. You can see the install.log starting with "Installing plugin from https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip"

kb-, 2020-08-08:
I've been able to install it after doing "uninstall & clean up..." on the previous version. I think it wouldn't install because this version still has the 0.0.7 number.

"Successfully installed Consolidated-Tabs-0.0.7Done!"

It's now working !

jneilliii, 2020-08-08:
Good point. I didn't think of that. I typically like to keep the version the same until official release. I probably will be doing some additional tweaking like if there aren't any sizes just make the panels full width by default instead of the 31% I'm using now.

kb-, 2020-08-08:
"Use Full Width of Browser" should only affect the tab containing Consolidated Tabs. The GCode Viewer tab doesn't look right with the full width.

jneilliii, 2020-08-08:
Yeah, I've been thinking of how to adjust for that. Currently the only way to be able to get the full width on the consolidated tab is to update the full width of the whole tabs container element. Otherwise there are incosistencies with the top tabs names and the line that goes underneath those links.

LarryVand, 2020-08-09:
Great job by the way. And the benefit of having a full width useable page, IMO, outweigh the GCode Viewer not using that full width.  

BTW, I don't know if this is a bug, but this is what happened when I first installed the consolidated tabs plugin.
I tried to size the Temperature and Control in the octoprint tab, and when I would size the Temperature to the right, Control would move in a proportional amount to the right by leaving an empty space gap. And when I would try and size the Control to the left, it would shrink Temperature to the left. It's like the consolidated tab window has a solid wall to the left and nothing on the right, as the control tab size would stay as is and be pushed far out of view to the right by sizing the temperature window, but the temperature window on the left will auto-shrink when I tried to size the control window.

And maybe bug #2? I then loaded DragOnOrder plugin to make the OctoPrint tab the first one (it was last when I first installed ConsolidatedTabs), and after it restarted Octoprint, the Temperature and Control consolidated tab were overlapping each other. I corrected them and then restarted octoprint to make sure the problem was fixed, and it was.

My suggestion is to allow the user to move and size without having these automatic moves to try and prevent overlapping. Also allow the tab that is selected to be the one in front (if the whole point of these automatic moves is always to show the "move button" that would normally be obscured if overlapping was allowed). IMO it will be much more user friendly rather than these automatic moves when you are setting up the tabs.  

Just my thoughts. And again great work BTW. Being able to use the full width with useable info is awesome! You are a star!

jneilliii, 2020-08-09:
The moving is something I'm not sure I'll be able to control. All of that is done as part of the jQuery UI draggable and resizable modules. I did attempt to resolve the issue in regard to bringing the panel to the front, but for some reason that broke at some point during development. When it was working you could click on the panel header (not the name) and it would bring it to the front. I'm sure there is some random CSS trick that I'm not familiar with that would allow all this stuff to collapse, I just haven't found it yet.

jneilliii, 2020-08-13:
What are your thoughts on removing the draggable capability all together and just make the panels sizable? All the panels would then just flow in the order they are in the plugin's settings, wrapping once they go beyond the tab container.

LarryVand, 2020-08-13:
Sounds like a good compromise. I will also be happy to beta test such a version, if you need some testing before release.

LarryVand, 2020-08-16:
I'm loving the new 0.0.8 version. Great job!!

jneilliii, 2020-08-16:
Thanks, it's been a struggle. There are still some quirks with panels locked to other panels and getting moved while resizing. I'm thinking of canning the whole thing though and going to something that supports a fixed grid layout and is more responsive. I have the initial test code working, but still want to make it better of course, but here's a preview.

[image]

LarryVand, 2020-08-16:
None of my panels did that with the new version (meaning being locked to other panels and moving when resized). All I know is 0.0.8 works beautifully and it is very user friendly. Make it even more user friendly by replacing "For the best experience with positioning and ordering of the tabs please check these Tips." with "Shift-Click to select a panel; then you can size it or click & drag panel header to move it; **see more tips here**".

jneilliii, 2020-08-19:
Just released 0.0.9 version that adds additional description in the settings on how to activate a panel for moving and resizing. If any of you guys are still having issues with version 0.0.8+ please open a new issue.

closed 2020-08-19 by jneilliii