jneilliii/OctoPrint-ConsolidatedTabs #10: How do I resize panels?

DesiQuintans, 2020-08-09:
It is not immediately obvious to me how to resize panels, nor is it not documented anywhere. I only get the dragging/moving widget at the top of each panel, and none of the other usual resizing verbs work (dragging the boundaries of the panel, dragging the corners). My settings are:

[image]

and the outcome is:

[image]

So it is also ignoring my 'overall width' setting. I have tried both Chrome and Firefox, same issue in both. Thanks!

jneilliii, 2020-08-09:
The plugin is being actively developed. There is a current branch that has made some tweaks to this. You can install by removing the current install and then use this URL in Plugin Manager > Get More > ...from URL. 

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

But to answer your question, the left,right,and bottom boundaries of the panels are resizable handles, click on those edges and hold down and drag.

corgan2222, 2020-08-19:
Same for me. My Panels are not side by side, but among themselves. And i also cant change the size. 
I have installed via the plugin manager and it says allready v0.0.8. Is the zip file you linked here different?

jneilliii, 2020-08-19:
Documented on the release notes and readme,  hold down your shift key and click a panel to activate drag/resize capabilities.

jneilliii, 2020-08-19:
Just updated plugin to 0.0.9 with more description in the settings on how to activate a tab for dragging/resizing.

closed 2020-08-19 by jneilliii