jneilliii/OctoPrint-ConsolidatedTabs #11: Improvement Request - Align main form to Left

EagleB3, 2020-08-12:
I'm not sure if I understand correctly the reasons for this, but, in my opinion, the necessary centering of the visual form on the screen interferes with optimal work.
Can I disable centering (or enable left alignment) in the plugin settings?

Hope the screenshots will explain my thoughts and wishes ...
[image]

[image]

jneilliii, 2020-08-12:
Please check out #3 and the current Feb version linked in that issue, it changes the option to full browser width, rather than specifying and has fixed several people's tab drop down issue.

jneilliii, 2020-08-12:
Basically, uninstall the existing version you have and use the URL below in Plugin Manager > Get More > ...fromURL and click Install.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

Once installed open settings and enable the option for using the full browser width.

EagleB3, 2020-08-12:
Many thanks!

Removed plugin, installed version 0.0.8 from ZIP file. After several experiments, at the moment when "resize" was disabled in the plugin settings, and "use full width" was disabled in the plugin settings, I got the following problem:
[image]

Now "Settings" button is closed by the "Select Extruder" button and is not available at any screen scaling. Resize is disabled in the plugin settings. The item for moving the bookmark is not displayed. The position of the panel cannot be changed and the plugin settings cannot be changed. 

Stalemate situation... :)

I managed to get out of this situation by loading octoprint in "safe mode" and clearing the plugin settings.

P.S. I use Firefox last actual version, 79.0 x64 on Windows 10.

EagleB3, 2020-08-12:
Continuing my research, I enabled the "resize" and "use the full width of the browser" option. I played a little more with the width and placement of the windows and got this picture (Scale=100%):
Left part
[image]
Scroll to right
[image]

It is EXCELLENT! It's AMAZINGLY CONVENIENT! Excellent width of the panels on the left, large font (scale = 100%), while you can see the main content of the line in the terminal, and if necessary, by simply scrolling the window to the right, you can see the entire line of the terminal. SUMPTUOUSLY!
I would really like to be able TO FIX THE CURRENT POSITION of the bookmarks in this form and USE IT IN FUTERE!

But, nevertheless, I'm afraid that this is a bug ... :)
This is how the window looks when zoomed out (scale = 60%):
[image]

Please, PLEASE think - how can you create (reproduce) such a position of bookmarks by purposeful plugin settings?

EagleB3, 2020-08-12:
Additional screenshots at different scales:
120% (Left)
[image]
120% (scroll right and slightly down )
[image]
110%
[image]

100% screenshot in the previous message

90%
[image]
80%
[image]
70%
[image]

60% - screenshot in the previous message

jneilliii, 2020-08-12:
I appreciate all the screenshots and will test with zoom scaling in the browser. Until I can find someone better with CSS though, we might be stuck.

jneilliii, 2020-08-12:
I just pushed an update to that same previous 0.0.8 one that might fix this issue. I need to verify in firefox, but in Chrome it does seem to scale fine. If you do want to test, please do the uninstall with the option to clear the data and start over fresh with the same URL below.
```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

jneilliii, 2020-08-12:
to be clear, in Chrome if you scale larger than 100% the panels will wrap down the page.

jneilliii, 2020-08-13:
What are your thoughts on removing the draggable capability all together and just make the panels sizable? All the panels would then just flow in the order they are in the plugin's settings, wrapping once they go beyond the tab container.

EagleB3, 2020-08-13:
> to be clear, in Chrome if you scale larger than 100% the panels will wrap down the page.

Just out of interest, without changing anything in the settings, I closed Firefox and connected through Chrome (84.0.4147.125 (Official build), (64 bit)). I see exactly the same position of the tabs on the screen, and the shadows are scaled exactly the same. Wrap down occurs at a scale> 110%.
I guess firefox allowed the tabs to be moved to the wrong position and saved the settings, and Chrome neatly reproduces the position according to the settings.

110%
[image]

100%
[image]

50%
[image]

jneilliii, 2020-08-13:
Thanks. I think removing the draggable option and forcing the wrap instead would resolve this issue, but awaiting a little more feedback from other users. So far, they seem like the compromise is acceptable. Ordering in the plugin's settings, and then sizing in the tab only. I pushed updates to the 0.0.8 version that may help already, by making only a selected tab resizable. I plan also to add a `lock panels` option that will disable the resizing and dragging (if we keep it) to avoid accidental moves. If you want to test it out, you can uninstall and use the same URL as before to install.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

I appreciate the feedback and testing.

EagleB3, 2020-08-13:
Sorry, I don't have time to write and check the translation. :)
I'm not fluent in English.

Yes, I saw your message and will definitely try the new assembly v.0.0.8.
 
> What are your thoughts on removing the draggable capability all together and just make the panels sizable? All the panels would then just flow in the order they are in the plugin's settings, wrapping once they go beyond the tab container.

For me personally, this option would be acceptable. Especially if it will be possible to set the width (and height!) of each bookmark.
But you need to think in advance about the question - how will the subsequent bookmarks turn around, if one of the previous ones is abnormally high? If the idea is not clear, then please write and I will attach an illustration.

And I'm afraid my opinion will not be very representative. In fact, I only use 4 tabs in my work. And one of these four (Bed Visualizer) I use once a week, it does not need to be consolidated for me. In total, I personally only need to consolidate three tabs. In the thread devoted to checkout #3 , I saw screenshots of users who have **a lot of tabs** (!); I'm afraid this option will seem bad to them.
The drag-and-drop option will probably allow you to create "half-height" bookmarks and place them on top of each other; users will probably take it very positively.

EagleB3, 2020-08-13:
> The drag-and-drop option will probably allow you to create "half-height" bookmarks and place them on top of each other; users will probably take it very positively.
If you allow users to create, for example, such designs with movable bookmarks of an arbitrarily customizable size, then, you must agree, it will impress them!
[image]

...If, of course, it is possible at all. But it all starts with a dream! :)

EagleB3, 2020-08-13:
I downloaded a new file, OctoPrint-ConsolidatedTabs-0.0.8
[image]
the archive size is different from the previous one, but the interface appearance and behavior in Firefox have not changed.

I don't know if this is the kind of feedback you expect, :)
but I seem to understand the order of the effect and its behavior.

As far as I understand, the offset of the tab (n my screenshot = "Terminal") positioning point ("Cross", but this is not essential) snaps to one of the bottom corners (bottom right corner?) of the tab, which is located to the left of the moved one (on my screenshot = "Control"). At some point in time, the combination of displacement and scaling moves the "Cross" out of the right border of the scaled area ("Border"). And this tab is excluded from scaling calculations and warp down execution.
[image]

Scaling will be performed for tabs that remain within the border, and the vacated tab will remain in place, while the "Cross" tab to which it is anchored remains within the "Border" (f.e.110%):
[image]
The next scaling step (120%) will warp down
[image]
but the released tab will not be tucked, it will fly far to the side
[image]

Now let's go back to the moment when the tab got rid of its neighbor to the left. By zooming out (scale = 50%), it can be found on the screen. Playing with its size and scale, you can see its upper border on the screen and lower it to the desired height. And also in this position (important! **ONLY** in this position), you can move its **RIGHT** border:
[image]
After the top frame of the free tab is lowered to the desired height, you can zoom in. The free tab will overlap with the neighbor to the left.
In this position, a free tab cannot be moved (its "Cross" is not available), but its right border can be moved. By scrolling the form by scrolling to the right, you can view the entire free tab. And you move it so that the neighbor on the left is fully visible. BINGO!
[image]
_If its width is not enough, then you can reduce the scale (scrуenshot "scale 50%") and drag its **RIGHT** border again, adjusting the total width of the free tab._

With this position of the tabs, everything will be perfectly scaled and scrolled to the right, as long as the "Cross" of the free tab is to the right of the "Border" of the scaled area (in my screenshot = 90%):
[image]
[image]

At the next step of zooming out (80%), "Cross" should fall within the border and the free tab will again fly far to the right:
[image]
[image]

I hope you find this research useful.
For example, before saving the new position of the moved tab, you can check for the condition that the coordinate is within the scaled region.

But I AM VERY asking YOU: leave the opportunity to turn off this control and make such a free tab.
This tab is amazingly handy! Delightfully comfortable!
Well, don't visualize this disabling of controls in the interface, but let it be modified / disabled with a text editor in the consolidatedtabs_settings.jinja2 file.

Or maybe, while debugging is in progress, try to add the plugin settings for each tab separately to disable (disable) the execution of warp down when scaling?

EagleB3, 2020-08-17:
Just installed the official update 0.8.8 - everything works great. Checked in Chrome and Firefox.
It's a great idea to activate the move / resize window with Shift + Click!

jneilliii, 2020-08-17:
Cool, can I close this issue now?

EagleB3, 2020-08-17:
Yes.

Many thanks!

closed 2020-08-17 by jneilliii