jneilliii/OctoPrint-ConsolidatedTabs #12: Tab layout issue when using monitors with different resolution

nforceroh, 2020-08-12:
I've got multiple monitors, 1920x1080 and my main monitor is 2k, 2560x1440, when you drag the browser between screens the tabs get all messed up and are no longer aligned
How it looks on 1920x1080
[image]
How it look when moved to 2560x1440 screen
[image]
Now set up for 2k screen moved to 1920x1080
[image]
to 1920x1080
[image]

jneilliii, 2020-08-12:
Yeah, this is a tricky one. Can you reproduce the same issue by uninstalling the plugin and installing my dev version using the link below in Plugin Manager > Get More > ...fromURL and clicking Install? After install go into settings and check the box for full width.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.0.8.zip
```

nforceroh, 2020-08-12:
So with the new version, it keeps the default layout correct between different resolutions. Changes in the horizontal axis such as width of the frames are kept but I can't drag the temperature below the controls like before. I can't make it go down more than this for some reason
[image]

jneilliii, 2020-08-12:
So with the new version there is a perimeter at the bottom that can be resized down to give you more room to move stuff. I think you might not be seeing it because of the Themeify you have because I don't see the actual line like you do in the default octoprint view. Not sure if that will prevent you from resizing it or not. But ideally if you want things to be ordered a specific way, order them in the plugin's settings first and then move/resize as necessary. The wrapping happens left to right, top to bottom.

jneilliii, 2020-08-12:
And Yes, I know this is real finicky and not ideal, but it's the best I got right now until some CSS guru jumps in and helps out.

nforceroh, 2020-08-12:
Ok thanks for the feedback, the other thing I see is weird, I can only drag down to resize like 10 pixels at a time or so, once I dragged the bottom all the way down, it will stack the temp and controls. I'll disable themeify later and see if it's different
here's a video i made of the issue for you . Don't mind the pecking around after 50s, stupid game bar video popup was there and I couldn't grab the frame move :) https://www.youtube.com/watch?v=Xe3GYfNx_RI
Just wanted you to be aware, you can close the issue

jneilliii, 2020-08-12:
Yeah, I was seeing the same thing, even though I can't see your YouTube video since it's private, but I know what you mean. I'll see if there is anything I can do to resolve that.

jneilliii, 2020-08-19:
Just released 0.0.9 version that adds additional description in the settings on how to activate a panel for moving and resizing. If any of you guys are still having issues with version 0.0.8+ please open a new issue.

closed 2020-08-19 by jneilliii