jneilliii/OctoPrint-ConsolidatedTabs #14: layout problem with TouchUI

n2hpz, 2020-08-16:
I use OcotPrint on both my PC monitor and a 3.5 inch screen on my printer.  I love the layout capabilities on my PC, but as soon I set up multiple tabs on my PC it makes the 3.5 inch screen useless.  The 3.5 inch screen is running TouchUI.  It would be great if there was a way to disable Consolidated Tabs for the small screen.

jneilliii, 2020-08-16:
I can definitely do that. I've had to deal with TouchUI before and disabling stuff when that was loaded and enabled. I thought that was in this plugin already, but I must have not done that.

n2hpz, 2020-08-25:
I just tried again.  Even did a total reboot of the pi.  No luck.

n2hpz, 2020-08-25:
Just to confirm, I shrunk my pc browser window as narrow as it would go and refreshed the webpage.  It switched to the TouchUI layout with the same error.
[image]

jneilliii, 2020-08-25:
Thanks, that's one thing I didn't try. What is the resolution of your pi's touch screen?

jneilliii, 2020-08-25:
I think I got it now. Was able to reproduce by mimicking a hyperpixel display in developer console. If you reinstall with that same URL above it should work better.

[image]

n2hpz, 2020-08-25:
that did it!!!!!!!!!!!!!!!

jneilliii, 2020-08-25:
Awesome. I just made a couple of more tweaks and will push an official version release now that you've confirmed. Thanks for the testing.

closed 2020-08-25 by jneilliii