jneilliii/OctoPrint-ConsolidatedTabs #17: Tabs missing resize or move controls

captdistraction, 2020-08-17:
I've been unable to resize consolidated tabs, I've tried limiting which plugins are installed, haven't spent much time in the browser consoles to see if there's a specific error yet, but wanted to make sure I wasn't missing anything simple. I had themify installed, but removed it in case it was preventing the display of those controls/CSS

[image]

installed plugins
[image]

jneilliii, 2020-08-17:
Hold down the shift key and click on a panel. That activates the panel and allows moving and resizing, as documented in the release notes, and in Tips on the readme. I will be adding it to the settings page as well in the next release and on the official plugin repository listing soon to avoid further confusion.

captdistraction, 2020-08-18:
confirmed, I'm an idiot user :) Thanks for the fast response

closed 2020-08-18 by jneilliii