jneilliii/OctoPrint-ConsolidatedTabs #23: Shift + click no longer allows me to drag the pannel

herbzhao, 2020-08-28:
I am using the 0.12 version..

jneilliii, 2020-08-28:
Do you have any errors in your browser's developer console (f12 and look for the console tab)? I'm not having any issues with the latest version 0.1.2.

herbzhao, 2020-08-29:
Hi, good idea. here is the error output:

packed_core.js?3fb354c1:833 Error calling onAllBound on view model ConsolidatedtabsViewModel : TypeError: this._addClass is not a function
    at $.<computed>.<computed>._create (http://octopi.local/static/webassets/packed_plugins.js?68308b70:253:23605)
    at $.<computed>.<computed>._create (http://octopi.local/static/webassets/packed_plugins.js?68308b70:1695:33)
    at $.<computed>.<computed>._createWidget (http://octopi.local/static/webassets/packed_plugins.js?68308b70:1865:16)
    at new $.<computed>.<computed> (http://octopi.local/static/webassets/packed_plugins.js?68308b70:1656:17)
    at HTMLDivElement.<anonymous> (http://octopi.local/static/webassets/packed_plugins.js?68308b70:1811:38)
    at Function.each (http://octopi.local/static/webassets/packed_libs.js?df208d52:1:2573)
    at w.fn.init.each (http://octopi.local/static/webassets/packed_libs.js?df208d52:1:1240)
    at w.fn.init.$.fn.<computed> [as selectable] (http://octopi.local/static/webassets/packed_plugins.js?68308b70:1803:17)
    at ConsolidatedtabsViewModel.self.onAllBound (http://octopi.local/static/webassets/packed_plugins.js?68308b70:1111:61)
    at callViewModelIf (http://octopi.local/static/webassets/packed_core.js?3fb354c1:832:75)
    at http://octopi.local/static/webassets/packed_core.js?3fb354c1:826:138
    at Pn (http://octopi.local/static/webassets/packed_libs.js?df208d52:1:92204)
    at Function.<anonymous> (http://octopi.local/static/webassets/packed_libs.js?df208d52:1:102635)
    at callViewModelsIf (http://octopi.local/static/webassets/packed_core.js?3fb354c1:826:95)
    at callViewModels (http://octopi.local/static/webassets/packed_core.js?3fb354c1:825:56)
    at bindViewModels (http://octopi.local/static/webassets/packed_core.js?3fb354c1:881:89)
    at sentryWrapped (http://octopi.local/static/webassets/packed_core.js?3fb354c1:430:45795)

jneilliii, 2020-08-29:
Thanks, that helps a little bit. Can you please provide a list of plugins you have installed that include a tab?

jneilliii, 2020-08-29:
I may have to get you to click on some of those `packed_plugins.js` lines too in order to get to the actual line of code that's causing the error, because your combined packed_plugins.js file will be different to mine.

herbzhao, 2020-08-29:
[image]
[image]

jneilliii, 2020-08-29:
Ok, I've installed each of these plugins one at a time and am not able to reproduce this error. If you can refresh your page usin shift+f5 on your keyboard and get the error again in developer console and then download the file http://octopi.local/static/webassets/packed_plugins.js and share both a screenshot of the error and the js file (I think you can drag the file into a comment) here that might help debug this more.

jup2000, 2020-09-13:
After installing Control Editor, it broke the shift click as well.  Not sure if it was related to this or upgrading to python 3

jneilliii, 2020-09-13:
I run consolidated tabs in Python 3 personally so don't think that's it. It's possible control editor has a conflicting draggable/sortable library that causes this one to break. I would recommend setting up your custom controls and saving, and then once saved disable that plugin. You will retain your custom controls with the plugin disabled. There is a known bug with that plugin that cause all your customizations to get wiped out anyway when it's enabled for what it's worth.

jneilliii, 2020-11-21:
Since the issue stems from the other plugin's bundled sortable library being older there's nothing I can do on this end to resolve this issue without great effort. As I mentioned before, you can create your custom controls and then disable that plugin and your controls will still be available and then you won't have the conflicting library loading.

closed 2020-11-21 by jneilliii