jneilliii/OctoPrint-ConsolidatedTabs #24: Tab Panel Colours Not Obeying Themify

BobaDog, 2020-09-14:
Hi,

Firstly, I really like your plugin.

However, I am having trouble getting the panels to obey the Themify settings. Please see images attached. 

I think I have added the correct CSS selectors but maybe not.

Can you help me please?

Thanks

[image]
[image]

jneilliii, 2020-09-14:
check the box enable customizations.

BobaDog, 2020-09-14:
Thanks for the prompt reply. My apologies. It was not checked in the image but it doesn’t make a difference when checked. Panels are still white. Thanks. 
    	
    	

    	Get Outlook for iOS

jneilliii, 2020-09-14:
Here are my settings and they work for me.
```
    themeify:
        customRules:
        -   enabled: false
            rule: background-color
            selector: .navbar-inner
            value: '#2f3136'
        -   enabled: false
            rule: background-color
            selector: .accordion
            value: '#2f3136'
        -   enabled: true
            rule: width
            selector: div.row-fluid.inline_thumbnail > img
            value: 100px
        -   enabled: true
            rule: display
            selector: div.row-fluid.inline_thumbnail > img
            value: block
        -   enabled: true
            rule: margin-left
            selector: div.row-fluid.inline_thumbnail > img
            value: auto
        -   enabled: true
            rule: margin-right
            selector: div.row-fluid.inline_thumbnail > img
            value: auto
        -   enabled: true
            rule: height
            selector: '#files > div > div.gcode_files > div.scroll-wrapper'
            value: 800px
        -   enabled: true
            rule: background-color
            selector: div.panel-heading
            value: '#2f3136'
        -   enabled: true
            rule: border-bottom
            selector: div.panel-heading
            value: '1px solid #fc8003'
        -   enabled: true
            rule: border
            selector: div.panel
            value: '1px solid #fc8003'
        -   enabled: true
            rule: background-color
            selector: div.panel
            value: '#2f3136'
        -   enabled: true
            rule: background-color
            selector: '#tabs_content'
            value: '#36393f'
        -   enabled: true
            rule: -webkit-box-shadow
            selector: '#tabs_content'
            value: unset
        -   enabled: true
            rule: box-shadow
            selector: '#tabs_content'
            value: unset
        -   enabled: true
            rule: border
            selector: '#sidebar'
            value: '1px solid #fc8003'
        -   enabled: true
            rule: border-top
            selector: .accordion-inner, .accordion-heading:gt(1)
            value: '1px solid #fc8003'
        -   enabled: true
            rule: border-radius
            selector: .accordion
            value: 4px
        -   enabled: true
            rule: background-color
            selector: .table thead tr.info > th, .table tbody tr td.info, .table tfoot
                tr.info > th
            value: '#36393f'
        -   enabled: true
            rule: color
            selector: div.x-axis-label
            value: '#dadadc'
        -   enabled: true
            rule: border-bottom
            selector: 'hr, #control h1'
            value: '1px solid #FC8003'
        -   enabled: true
            rule: border-top
            selector: hr
            value: '0'
        enableCustomization: true
```

[image]

BobaDog, 2020-09-14:
Thanks. I’ll try them tomorrow. I think I may of seen where I went wrong. I didn’t put the #FC8003 after 1px solid. I couldn’t see it on the image on your site. Fingers crossed. 
Thanks for the prompt replies. 
    	
    	

    	Get Outlook for iOS

closed 2021-01-31 by github-actions[bot]