jneilliii/OctoPrint-ConsolidatedTabs #26: Multicam plugin not able to switch between cameras

Jarren141, 2020-09-17:
Your plugin is awesome but there is this one bug that could be improved. 

The plugin only switches views if I switch tabs on chrome and come back to octoprint but reloading the page with a different camera results in the default webcam being displayed.

The Plugin manager shows
"Consolidated Tabs (0.1.2)
 Homepage  jneilliii
Combines configured tabs into a single tab.
 AGPLv3  Python >=2.7,<4  "

[image]
[image]

Jarren141, 2020-09-17:
Google Chrome Version 85.0.4183.102 (Official Build) (64-bit)

jneilliii, 2020-09-18:
Duplicate of #20 I submitted a PR to that plugin's repo that will resolve this issue for you. See the details in the closed issue.

closed 2020-09-18 by jneilliii

jneilliii, 2020-09-18:
I just checked and my changes were merged within that plugin repo, but there was not a new version released. You should be able to uninstall the plugin keeping the settings and reinstall the multicam plugin and get the newer version that should work with this plugin.

Jarren141, 2020-09-21:
Thanks so much! Sorry to make a duplicate. this was my first ever "issues" thing. Totally fixed the issue!