jneilliii/OctoPrint-ConsolidatedTabs #27: Consolidated Tabs and Widescreen plugins don't "play nice" together

theone11, 2020-09-27:
When I install both the the tabs go under the left sidebar and under them the right side sidebar.

**Expected**:
Left SideBar       Tabs         Right SideBar

**Actual**:
Left SideBar
Tabs
Right SideBar

Can this be fixed so both plugins can work together as expected?

jneilliii, 2020-09-27:
At one point during this plugin's life this did work without issue. It was probably messed up when I switched to full screen option instead of set width option. I'll have to review again.

markeby, 2020-10-25:
I am using both without any problems on octoprint version 1.4.2 and updated plugins.

jneilliii, 2020-11-21:
@theone11 any update on your side in regard to this issue? As @markeby mentioned it works fine for him too, so I will need more details from you to move this forward for the next release.

theone11, 2020-11-29:
If I zoom out the browser eventually the view is OK (side-by-side)
I tried unmarking "Use full width of Browser" and now it shows correctly in regular 100% zoom.

jneilliii, 2021-03-06:
My latest gridstack development version of this plugin might resolve this issue. If you want to test yourself you can install by copying/pasting the URL below into Plugin Manager > Get More > ...from URL and clicking install. You will have to re-add your tabs  after the initial install.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

closed 2021-03-07 by jneilliii