jneilliii/OctoPrint-ConsolidatedTabs #28: Dark mode support

NovaViper, 2020-09-29:
Hi, Im using the plugin along with Themify and I notice that the items within the Consolidated tabs have a white background and do not match up with the themes within Themify. I'm using Octoprint 1.4.2
[image]

jneilliii, 2020-09-29:
Read the docs. There are pointer on how to do this in Themeify's advanced section and is how I do it...

jneilliii, 2020-09-29:
Specifically here...
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs#themeify

jneilliii, 2020-09-29:
and here's an exert from my Themeify settings for reference.
```
        -   enabled: true
            rule: background-color
            selector: div.panel-heading
            value: '#2f3136'
        -   enabled: true
            rule: border-bottom
            selector: div.panel-heading
            value: '1px solid #fc8003'
        -   enabled: true
            rule: border
            selector: div.panel
            value: '1px solid #fc8003'
        -   enabled: true
            rule: background-color
            selector: div.panel
            value: '#2f3136'
```

closed 2020-09-29 by jneilliii

NovaViper, 2020-10-01:
Ah, cool. Thanks!