jneilliii/OctoPrint-ConsolidatedTabs #29: Terminal Contents not selectable when terminal on Consolidated Tab

TodWulff, 2020-10-12:
There doesn't seem to be any workaround for terminal contents not being selectable when the terminal is configured to reside on the consolidated tab?  When it is on it's native tab, content can be selected and copied therefrom.  If terminal is an element on the consolidated tab, when one tries to select terminal content, one gets a dashed line UI element (think window selection) that doesn't make any clipboard changes when the mouse button is released.

https://i.imgur.com/aje0Bvm.png <-- on consolidated tab
https://i.imgur.com/B6avEiU.png  <-- on native tab

jneilliii, 2020-10-15:
I've just released an 0.1.3rc1 version that should resolve this for you. If you'd like to try it out and verify you can copy/paste the URL below into Plugin Manager > Get More > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/0.1.3rc1.zip
```

TodWulff, 2020-10-15:
Installed.  Testing now.  Thank you, kind sir.

-t

TodWulff, 2020-10-15:
https://i.imgur.com/8S7MSat.png

Issue is resolved with this version.  Again, thank you.  Much appreciated.

closed 2020-10-15 by TodWulff