jneilliii/OctoPrint-ConsolidatedTabs #32: Nothing happen when I 'Shift + click'

scott0122, 2020-11-07:
OctoPrint 1.4.2
Python 3.7
Chrome Version 86.0.4240.183 (Official Build) (64-bit)

Error occured while connect to OctoPring 

```
packed_core.js?411b1f0f:826 Error calling onAllBound on view model ConsolidatedtabsViewModel : TypeError: this._addClass is not a function
    at $.<computed>.<computed>._create (http://midi-octo-01:5000/static/webassets/packed_plugins.js?255a8e5d:11:23605)
    at $.<computed>.<computed>._create (http://midi-octo-01:5000/static/webassets/packed_plugins.js?255a8e5d:1453:33)
    at $.<computed>.<computed>._createWidget (http://midi-octo-01:5000/static/webassets/packed_plugins.js?255a8e5d:1623:16)
    at new $.<computed>.<computed> (http://midi-octo-01:5000/static/webassets/packed_plugins.js?255a8e5d:1414:17)
    at HTMLDivElement.<anonymous> (http://midi-octo-01:5000/static/webassets/packed_plugins.js?255a8e5d:1569:38)
    at Function.each (http://midi-octo-01:5000/static/webassets/packed_libs.js?df208d52:1:2573)
    at w.fn.init.each (http://midi-octo-01:5000/static/webassets/packed_libs.js?df208d52:1:1240)
    at w.fn.init.$.fn.<computed> [as selectable] (http://midi-octo-01:5000/static/webassets/packed_plugins.js?255a8e5d:1561:17)
    at ConsolidatedtabsViewModel.self.onAllBound (http://midi-octo-01:5000/static/webassets/packed_plugins.js?255a8e5d:869:61)
    at callViewModelIf (http://midi-octo-01:5000/static/webassets/packed_core.js?411b1f0f:825:75)
```

[image]

jneilliii, 2020-11-07:
Does the same thing happen with custom controls disabled? You shouldn't lose your custom controls BTW with that plugin disabled, you'll just not be able to modify them.

jneilliii, 2020-11-07:
Just looked at custom control, and I bet there is a conflict as a result because it also adds the jqueryui sortable module, which in turn is breaking the one that I have included.

scott0122, 2020-11-07:
It works now after Custom Control Editor disabled, thanks.

jneilliii, 2020-11-07:
Cool, I've actually requested the full jQuery UI module be added to core OctoPrint in the above issue so all plugins can utilize its features and not step on each others toes (assuming we can get plugin author to remove the one bundled).

jneilliii, 2020-11-21:
duplicate #23

closed 2020-11-21 by jneilliii