jneilliii/OctoPrint-ConsolidatedTabs #36: [BUG]: Buttons are no longer compact

EagleB3, 2020-12-01:
**Describe the bug**

Before Octoprint was updated to version 1.5.0, the buttons on the tab were displayed very compactly and it was convenient.
Immediately after the update to 1.5.0, the buttons became huge and do not fit into the previously reserved area of the bookmark. Updating the plugin to versions 0.1.6 and 0.1.7rc2 did not solve this problem.

**Expected behavior**

I really liked the compact layout of the buttons, and I hoped that there would be no changes when changing the version, since the compact presentation of the buttons was very convenient to use.
[image]

_**Maybe it is possible to give the buttons a minimum width (by the size of "Icon + text") when specifying "Width = 0" in the settings?**_

**Screenshots**

I don't have the old screenshot left. All buttons were assigned "**Width = 2**" and on the tab they looked like this:
[image]

Immediately after the update, they turned into the following:
[image]

[image]

After the upgrade, the best I can achieve is:
[image]

**Browser errors**

There are no errors in the browser. There are warnings like "Expected media function name, but found 'touch-enabled'.". Not sure if this is relevant to the problem
[image]

jneilliii, 2020-12-01:
Should this have been posted in terminal commands extended plugin, not here?

EagleB3, 2020-12-01:
@jneilliii , Yes, I'm sorry. I'm a little confused.

closed 2020-12-01 by jneilliii