jneilliii/OctoPrint-ConsolidatedTabs #38: [BUG]: Consolidated Tabs interferes with the operation of Pretty GCode Plugin

sonjamichelle, 2020-12-11:
**Describe the bug**
When the Consolidated Tabs and Pretty GCode plugins are installed together the Pretty GCode Information panels do not open when their buttons are clicked.

**Expected behavior**
Click on panel activation button, panel is supposed to open.

**Screenshots**
[image]

[image]

**Browser errors**

https://www.designcreations.us/images/pg_1.png

https://www.designcreations.us/images/pg_2.png

My report on Pretty GCodes GitHub (https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/80#issuecomment-743152548)

Sorry it has taken me so long to respond and report, been busy on the homefront.

I went and disabled EVERY 3rd party plugin save for PG, rebooted the system, and PG worked as expected.

I then began enabling my plugins one by one. When I got to Consolidated Tabs (https://github.com/jneilliii/OctoPrint-ConsolidatedTabs) PG began to ac up again. I disabled that Consolidated Tab, rebooted and PG worked as expected.

I then went through, enabled the rest of my 3rd party plugins save for Consolidated Tabs, in fact I completely removed it and PG continued to work as expected.

So it look like Consolidated Tabs is the culprit.

I also opened an issue on Consolidated Tabs GitHub

jneilliii, 2020-12-11:
The issue is that we're both manipulating the DOM. I'll see what pretty gcode is doing here, but I thought I already had added some tweaks to my plugin to resolve some of this. I don't use the fullscreen dashboard view so am unsure how that works currently.

jneilliii, 2021-03-22:
Do you know if this issue still exists for you in the latest 1.0.0+ versions that implements gridstack.

sonjamichelle, 2021-03-22:
I have both plugins installed at their latest versions. Everything seems to work as it should. One or the other developer made the necessary changes to get the plugins to cooperate with each other.  You'd have to look through the PR's/Commits to see the actual change.

closed 2021-03-22 by sonjamichelle