jneilliii/OctoPrint-ConsolidatedTabs #39: [FR]: Flex layout

jannessm, 2020-12-19:
Hi,
I have used your plugin for some time now and wondered why you haven't used the css flexlayout option? I have seen some commented code, so I was wondered why you haven't used it?

I would recommend using this, since there are a few problems with your current solution. When adjusting the height of the first box, all boxes on the right are moving, too. Sometimes this leads to boxes ranging out of the window. Then I am not able to move it back to the bottom again. A flex layout may solve this issue including boxes that overlap. Maybe even for multiple screensizes.

jneilliii, 2020-12-19:
Mainly because of my unfamiliarity with modern day css techniques. I started experimenting with them but still couldn't get it to work the way I wanted it to. If you can provide suggestions on how to get it to work or provide a PR that would be very welcome. My HTML/CSS knowledge is very dated.

jneilliii, 2020-12-28:
I have a development version available that I've been working on that utilizes gridstack.js which makes the sizing based on a defined grid. Adding tabs and editing is done from the button in the bottom right of the tab once installed. If you want to give it a try you can install using the URL below in plugin manager > get more > ...from URL and clicking Install. I think this is going to resolve a weird quirkiness with the resizing/dragging that is in the current version.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

EagleEyeBW, 2021-02-06:
@jneilliii this is quite simple
[image]
I just added this in the html code from my browser:
- add to the div that contains the tabs display:flex and flex-wrap: wrap
- and instead of the size of a tab set min and/or max width, or flex-grow or flex-shrink

here you can find some good infos https://css-tricks.com/snippets/css/a-guide-to-flexbox/

your gridstack.zip has some trouble for me:
- every first time I enter the settings its shrinkt to maybe 50px, after closing and reopen it works
- the tabs doesn't fit into the container

[image]

jneilliii, 2021-02-06:
Are you using UI Customizer in conjunction with this plugin? That seems to have this effect of not allowing the resizers to be there and I think the change I made to try and work around it didn't help and caused the settings thing. Could you test with this link? I don't want to put work into the flex stuff if it's just going to go away with the gridstack stuff, because the gridstack implementation resolves a bunch of stuff I didn't like about my own.

https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/7be0a7566e8c34db6991959ab102198fd40b5999.zip

EagleEyeBW, 2021-02-06:
no I just have themeify installed.
I would suggest to use as less js help as possible, my learnings over past years is. These stuff brings you in hell :D
Sure if there are importen fuctions that you need then there are valid points. But as long as there are standard ways to let the browser do the work, this is much more effective.

Jep this version works much better, but I would suggest to investigate in the usage flex layout stuff. 
I get some strage white spaces, but this may result of the alpha state of the plugin

jneilliii, 2021-02-06:
Strange whitespace? The gridstack implementation in the previous link doesn't auto-collapse things like the previous iteration, so you can technically drag the items anywhere you want to within the 24 column grid. Is that what you are seeing?

jneilliii, 2021-02-06:
and I appreciate the guidance here on modern day css...I'm old and don't these new HLTM5 capabilities.

EagleEyeBW, 2021-02-06:
this whitespace, even in the "Edit Mode" it exists, someone creates more height than needed
[image]

Absolutly no problem, I had to dig deep into this crap in the past years ;-)
Just as a small overview, if you have this HTML
```html
<div class="parent">
  <div class="child">
  <div class="next-child">
</div>
```

so if you whant to use flex layout, you can create this css, this let the first child Element increase its size and the second will not change the size.

```css
.parent {
  display: flex;
}
.child{
  flex-grow: 1;
}
.next-child{
  flex-grow: 0;
}
```

here you can see many of the options that the flex layout provides you:
https://css-tricks.com/snippets/css/a-guide-to-flexbox/

Everything is just done by css, no js. So you just have to create css or styling informations for the browser to modify all this.
So you can also modify the behaviour when its mobile oder other resolutions

jneilliii, 2021-02-06:
Yeah, that's the dragging part. You should be able to drag that panel up to the top. This was introduced when I disabled one of the options so that the panels wouldn't auto-collapse into whitespace, which is something I'm considering putting back in, or adding a button to collapse as much as possible. gridstack does provide those options built in.

EagleEyeBW, 2021-02-06:
ahhhh ok I got this wrong 👍

EagleEyeBW, 2021-02-06:
but I still have the issue that the settings dialog is on the first time
[image]

also I recognize, when I move the upper left corner out of the area it disappears

jneilliii, 2021-02-06:
yeah, if the panel gets dragged outside of the area it gets deleted. I'm contemplating some form of delete button instead to make that a little more reliable. I'll definitely look into the settings thing, thanks for the feedback.

closed 2021-02-28 by github-actions[bot]