jneilliii/OctoPrint-ConsolidatedTabs #40: [BUG]: Resizing panels messes up layout.

knoellle, 2020-12-27:
**Describe the bug**
When the size of a panel changes, it messes up the position (and sometimes size) of other panels even when they are not directly adjacent or attached to it.
It happens when the panel resizes itself as well as when manually resizing after shift-clicking the panel.
In this way, panels can even be moved up and off screen, which makes them quite difficult to recover.

**Expected behavior**
None of the panels that are not directly below the one changing size should be affected in any way.

**Screenshots**

Hard to show with screenshots, so here's a video:

https://user-images.githubusercontent.com/33729490/103161268-7029d180-47df-11eb-8b29-862a20a97365.mp4

**Browser errors**
None

jneilliii, 2020-12-27:
I'm aware of this issue and that is one of the reasons I made the video. It's best to position the tabs best you can in the plugin's settings first. This way it minimizes the impact of the resizing/positioning. I never could figure out a good way to have them truly float independently of each other.

knoellle, 2020-12-27:
Thanks for the quick reply.
If I understand correctly, the issue is that all panels are positioned relative to their predecessor in the list. So if one moves, all the ones beyond move as well.
I'm by no means an expert in html/css, but couldn't you just use absolute positions?

After some testing using [this](https://github.com/openstyles/stylus) browser extension, it does seem to me that setting the `position` attribute to `absolute` allows the panels to be positioned independently of each other.
I wasn't able to position the panels by shift-clicking and dragging however, since their `top` values are still set to large, negative numbers once moved.
When overriding those as well, I can - although tedious - position each panel without affecting any other panel.

jneilliii, 2020-12-28:
I have a development version available that I've been working on that utilizes gridstack.js which makes the sizing based on a defined grid. Adding tabs and editing is done from the button in the bottom right of the tab once installed. If you want to give it a try you can install using the URL below in plugin manager > get more > ...from URL and clicking Install. I think this is going to resolve a weird quirkiness with the resizing/dragging that is in the current version.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

jneilliii, 2021-03-22:
Do you know if this issue still exists for you in the latest 1.0.0+ versions that implements gridstack.

closed 2021-05-23 by github-actions[bot]