jneilliii/OctoPrint-ConsolidatedTabs #41: Cannot resize or move panels at all

markwinger, 2021-01-15:
Installed consoldated tabs.   After selecting the panels and use full screen, the panels include tabs bar appears below the state panel.  If I unselect use full screen, it moves to the right of the state panel.  In either case I cannot select the panels and os cannot resize of move them as shown in the video.

I am on a large 4k video monitor if that matters.
I have Dragon Order installed incase that is interfering.
Running Octoprint 1.5.2 on RPI 3b.
Had to uninstall for now.

jneilliii, 2021-01-15:
> I have Dragon Order installed incase that is interfering.

Yes, that probably is the culprit here. Try disabling that (it doesn't need to be enabled for the re-ordering to work, just when you need to re-order stuff).

Custom controls has a known bug that will cause them to get wiped out. I've had to deal with that before. You might want to install my patched version that prevents this by copying/pasting the URL below into Plugin Manager > Get More > ...from URL and clicking Install.

```
https://github.com/jneilliii/octoprint-customControl/archive/devel.zip
```

jneilliii, 2021-01-15:
oh...and to move/resize tabs once consolidated, you need to hold down the shift key when clicking on the tab...

jneilliii, 2021-01-15:
another option is to try my current dev branch, which uses a different javascript library for sizing/dragging. Copy/Paste the URL below into Plugin Manager > Get More > ...from URL and click Install. It basically puts the tabs into a 24 column width grid.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

markwinger, 2021-01-15:
I installed your dev branch and the config has  only 2 check boxes with no tab selections.
I reinstalled the the original version and was able to get it to work, using shift select. (I missed that in the video)
I have been saving custom controls fro the yaml file so I have not tried your patched version.

jneilliii, 2021-01-15:
Yeah, sorry. Forgot to mention the dev version uses a button on the tab in the bottom right. Click that button and you get plus signs on the existing tabs to click.

closed 2021-02-08 by github-actions[bot]