jneilliii/OctoPrint-ConsolidatedTabs #43: Consolidated Tabs doesn't work with Fullscreen Plugin

KonigTX, 2021-02-21:
Would love for the two plugins to play nicely. When going fullscreen with my webcam, the tabs that are consolidated do not hide and block view of the webcam. 

Fullscreen plugin: https://github.com/BillyBlaze/OctoPrint-FullScreen

Example of issue:
[image]
[image]

jneilliii, 2021-02-21:
I've been working on a different implementation of the consolidated tabs. Do you think you could test my current development version to see if has the same issue or not please? You can install by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking Install.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

KonigTX, 2021-02-21:
Will do once this print finishes!

KonigTX, 2021-02-21:
Love this version! So easy to size and move tabs now. Great work!

Unfortunately same issue still exists:
[image]

jneilliii, 2021-02-21:
There are options in the settings for that full width stuff. I find full width a little easier to handle when consolidating tabs personally. The gridstack implementation will also put all the panels as a single column width once the page gets resized to a certain width. 

As for the fullscreen plugin. I just patched it's css in my plugin to probably resolve this. You can reinstall using the same URL as before and report back please.

Thanks for the feedback on the development, that was the goal to make positioning and resizing better. I'm glad my efforts are headed in the right direction.

KonigTX, 2021-02-21:
It's absolutely a step in the right direction. Much more intuitive than previous implementation. Easier to get good results too.

Unfortunately the new version (0.1.4.dev11 to verify I'm on right ver) did not fix the issue. 

A couple wishlist items if you're interested:
- I don't mind the drag to remove panel, but I have destroyed my layout a few times already trying to do that. Maybe add a red trashcan delete button to top right of each consolidated tab while in edit mode? If you don't want to for aesthetics I understand
- Add an option to (temporarily) disable grid snapping? Don't get me wrong - I love grid snapping and you implemented it beautifully - but there have been a couple times I'd like to disable it for more precise sizing (one of the grid snaps left the top part of some text visible and it tweaked my inner pixel-perfectionist)
- Add a default set of grid options for fast easy config? 3x2, 2x3, 3x3 etc? This one probably isn't a big deal just a random thought while using the plugin

This is my new favorite plugin with this dev version. Seriously great work!

jneilliii, 2021-02-21:
Ok, I'll install that other plugin and see what css I need to apply to fix it. I thought for sure that would work. 

> Maybe add a red trashcan delete button to top right of each consolidated tab while in edit mode?

I was already thinking of doing this.

> Add an option to (temporarily) disable grid snapping?

Unfortunately the gridstack.js module I'm using doesn't have an option for this, so probably not going to be possible. I tried to account for it by making the "cell height" 25px instead of the same width as the columns as that seemed like a good compromise, but maybe an even smaller value like 10px would make more sense and allow a little more control. 

> Add a default set of grid options for fast easy config?

This would be possible, but from what I've seen there are some tabs that just don't play nice with certain sizes so figured it best to just let the user adjust.

> I'm sure you don't want/need more bug reports but if you try to click settings while in edit mode it loads the settings window in a tiny popup rendering it unusable.

I have seen this and thanks for pointing out that it was in edit mode. That was driving me crazy to figure out. I believe I know why this is happening but I also found if you close the window and reopen it should be normal size as well. You don't necessarily have to disable edit mode I don't think.

KonigTX, 2021-02-21:
You rock!

> Unfortunately the gridstack.js module I'm using doesn't have an option for this, so probably not going to be possible. I tried to account for it by making the "cell height" 25px instead of the same [...]
Maybe add an option for the value? I'm always a fan of user-defined options! As it is right now though -- if it would break something I wouldn't bother. It works very well and is a minor nuisance that is really _not_ a big deal.

Last thing I'll bug you with: if the width of a tab exceeds the set width of the consolidated tab could a horizontal scroll bar be added? If so could the plugin save the location of the horizontal scroll so it always loads to the same spot when the server is rebooted/reloaded/refreshed/etc?

jneilliii, 2021-02-21:
> Last thing I'll bug you with: if the width of a tab exceeds the set width of the consolidated tab could a horizontal scroll bar be added? If so could the plugin save the location of the horizontal scroll so it always loads to the same spot when the server is rebooted/reloaded/refreshed/etc?

That's something I'm not sure would be possible with gridstack either. The whole point of the implementation was to put some order to it. With the way I implemented (24 columns with 25px cell height) it seemed like the best compromise for sizing default tabs and adjustability, but I do like your idea of making that cell height a configurable option. I would consider making the column count configurable too, but because of gridstack's dependency on custom css for those column widths I fear that would be even harder to achieve.

jneilliii, 2021-02-21:
So I think the issue might be with your Themeify side. I just tried this other plugin and when I go into full screen mode I don't see this problem.

[image]

KonigTX, 2021-02-21:
> That's something I'm not sure would be possible with gridstack either. The whole point of the implementation was to put some order to it. With the way I implemented (24 columns with 25px cell [...]

No problem. I'm probably just nitpicking anyway.

>So I think the issue might be with your Themeify side. I just tried this other plugin and when I go into full screen mode I don't see this problem.

I don't have Themeify installed!
[image]

jneilliii, 2021-02-21:
I must be missing an option then, sorry. I thought the black bar at the top indicated a Themeify install. Is that something in the fullscreen webcam plugin?

jneilliii, 2021-02-21:
or is that a conflict with my floating navbar plugin...lol?

KonigTX, 2021-02-21:
> I must be missing an option then, sorry. I thought the black bar at the top indicated a Themeify install. Is that something in the fullscreen webcam plugin?

I thought it was Floating Navbar (0.3.4) but turns out you're the author of that plugin so you'd probably already know about it lol!

jneilliii, 2021-02-21:
yeah, just realized that might be the case. I wonder if that plugin is causing an additional layer of issue here, let me test with that one enabled.

KonigTX, 2021-02-21:
> yeah, just realized that might be the case. I wonder if that plugin is causing an additional layer of issue here, let me test with that one enabled.
Not sure if I'd bother... just disabled it and issue persists. Disabling all my plugins one-by-one to whittle this down to the culprit

jneilliii, 2021-02-21:
also force a refresh of your browser cache, just to make sure.

KonigTX, 2021-02-21:
> also force a refresh of your browser cache, just to make sure.

Done, no change

Disabled all 3rd party plugins except for the two conflicting, no change:
[image]

jneilliii, 2021-02-21:
so weird, what browser are you using?

KonigTX, 2021-02-21:
> so weird, what browser are you using?

Brave

jneilliii, 2021-02-21:
I'll have to try that one. I just pushed a release to Floating Navbar to fix that conflict (in theory). If that version works for you I might be able to implement the same patch in this one to resolve this problem. Could you please test that version and let me know if the navbar still shows or not?

KonigTX, 2021-02-21:
> I'll have to try that one. I just pushed a release to Floating Navbar to fix that conflict (in theory). If that version works for you I might be able to implement the same patch in this one to resolve this problem. Could you please test that version and let me know if the navbar still shows or not?

Just tried Edge, same issue, tried Internet Explorer and it wouldn't even load OctoPrint's UI so there's that.

Will try the new version of Floating Navbar when the update hits me

KonigTX, 2021-02-21:
> Just tried Edge, same issue, tried Internet Explorer and it wouldn't even load OctoPrint's UI so there's that.
>
> Will try the new version of Floating Navbar when the update hits me

0.3.6 didn't change anything

jneilliii, 2021-02-21:
I'm stumped at the moment. Both fixes worked perfectly for me, so I'm not sure the root cause of this for you. When you look at your developer console (typically f12 on your keyboard) and look at the elements tab while in fullscreen mode do you see this z-index for the navbar?

[image]

and this z-index for the webcam rotator?

[image]

KonigTX, 2021-02-22:
Same as you:

navbar:
[image]

webcam rotator:
[image]

jneilliii, 2021-02-22:
I think I have an idea. I'll let you know after I've had some time for additional testing.

jneilliii, 2021-02-22:
potential fix in same branch, version 0.1.4.dev13 available that I think may get us closer. can you give that URL a try again and let's see where we are.

KonigTX, 2021-02-22:
7 hours left on this print, will update it again tomorrow!

KonigTX, 2021-02-22:
It works with dev13!

What'd you do to fix it?

jneilliii, 2021-02-22:
I added some css rules to set z-index to inherit instead of specifying it as a set number for all the panels and their content when that plugin goes into fullscreen mode.

KonigTX, 2021-02-22:
> I added some css rules to set z-index to inherit instead of specifying it as a set number for all the panels and their content when that plugin goes into fullscreen mode.

Awesome! Thanks again for all your help. Keep up the great work with your plugins! 👍👍

closed 2021-03-07 by jneilliii