jneilliii/OctoPrint-ConsolidatedTabs #44: [BUG]: Cannot select / copy text from Octoklipper

Sineos, 2021-03-05:
**Describe the bug**
When using this plugin, I can no longer select or copy text from the octoklipper plugin window.

**Expected behavior**
Be able to select and copy text from the Octoklipper Message box

**Screenshots**
n/a

**Browser errors**
This is the html code of the octoklipper section:

<details> 
<summary>html</summary> 

```html
<div id="tab_plugin_klipper_main" class="" data-bind="allowBindings: true">
                                            <div class="row-fluid">
  <div class="span8">
    <label><i class="icon-tasks"></i> Messages</label>
    <div class="plugin-klipper-log" data-bind="foreach: logMessages">
      <div class="log-item info" data-bind="css: type">
         <div data-bind="text: time" class="ts">14:46:35</div>
         <div data-bind="html: msg" class="msg">Connected to host via /tmp/printer @250000bps</div>
      </div>
    
      <div class="log-item info" data-bind="css: type">
         <div data-bind="text: time" class="ts">14:46:35</div>
         <div data-bind="html: msg" class="msg">Firmware version: v0.9.1-294-gab979a52</div>
      </div>
    
      <div class="log-item info" data-bind="css: type">
         <div data-bind="text: time" class="ts">14:46:43</div>
         <div data-bind="html: msg" class="msg"> Klipper state: Ready<br></div>
      </div>
    
      <div class="log-item info" data-bind="css: type">
         <div data-bind="text: time" class="ts">15:15:08</div>
         <div data-bind="html: msg" class="msg"> Klipper state: Disconnect<br></div>
      </div>
    
      <div class="log-item info" data-bind="css: type">
         <div data-bind="text: time" class="ts">15:15:15</div>
         <div data-bind="html: msg" class="msg"> Klipper state: Ready<br> Klipper state: Ready<br></div>
      </div>
    
      <div class="log-item info" data-bind="css: type">
         <div data-bind="text: time" class="ts">15:50:58</div>
         <div data-bind="html: msg" class="msg"> Mesh Leveling Probed Z positions:<br> -0.042631 0.009417 0.026177 0.043872 0.059304 0.087043 0.084094 0.075918 0.072312<br> -0.008147 0.041361 0.045332 0.064297 0.054573 0.064107 0.087671 0.066954 0.079568<br> 0.019739 0.050631 0.060150 0.081948 0.080984 0.062647 0.068604 0.070399 0.056092<br> 0.039477 0.071684 0.079655 0.080546 0.084357 0.100810 0.088824 0.066443 0.070779<br> 0.049755 0.044777 0.088853 0.097452 0.076210 0.089496 0.080444 0.072750 0.077772<br> 0.051975 0.047303 0.061260 0.076385 0.094766 0.091204 0.075465 0.057727 0.046996<br> 0.055595 0.047785 0.079144 0.082824 0.102752 0.091613 0.091846 0.079874 0.064457<br> 0.055858 0.062968 0.082794 0.094897 0.101088 0.128593 0.114271 0.089831 0.086663<br> 0.092795 0.079626 0.106402 0.111074 0.132798 0.153792 0.161618 0.124184 0.101569<br></div>
      </div>
    </div>
    &nbsp;
    <button class="btn btn-mini pull-right clear-btn" data-bind="click: onClearLog, enable: isActive()" title="Clear Log">
       <i class="fa fa-trash"></i>Clear
    </button>
  </div>
  <div class="span4">
    <div class="control-group">
     <div class="control-group">
        <div class="controls">
           <label class="control-label"></label>
           <button class="btn btn-block btn-small" data-bind="click: onGetStatus, enable: isActive()" title="Query Klipper for its current status">
              <i class="fa icon-black fa-info-circle"></i> Get Status
           </button>
        </div>
     </div>
     <div class="control-group">
        <div class="controls">
          <label class="control-label small"><i class="icon-refresh"></i> Restart</label>
          <button class="btn btn-block btn-small" data-bind="click: onRestartHost, enable: isActive()" title="This will cause the host software to reload its config and perform an internal reset">
             Host
          </button>
          <button class="btn btn-block btn-small" data-bind="click: onRestartFirmware, enable: isActive()" title="Similar to a host restart, but also clears any error state from the micro-controller">
             Firmware
          </button>
        </div>
     </div>
     <div class="control-group">
        <div class="controls">
         <label class="control-label"><i class="icon-wrench"></i> Tools</label>
         <button class="btn btn-block btn-small" data-bind="click: showLevelingDialog, enable: isActive()" title="Assists in manually leveling your printbed by moving the head to a configurable set of positions in sequence.">
            Assisted Bed Leveling
         </button>
         <button class="btn btn-block btn-small" data-bind="click: showPidTuningDialog, enable: isActive()" title="Determines optimal PID parameters by heat cycling the hotend/bed.">
            PID Tuning
         </button>
         <button class="btn btn-block btn-small" data-bind="click: showOffsetDialog, enable: isActive()" title="Sets a offset for subsequent GCODE coordinates.">
            Coordinate Offset
         </button>
         <button class="btn btn-block btn-small" data-bind="click: showGraphDialog, enable: isActive()" title="Assists in debugging performance issues by analyzing the Klipper log files.">
            Performance Graph
         </button>
        </div>
     </div>
     <div class="controls">
         <label class="control-label"><i class="icon-list-alt"></i> Macros</label>
         <div data-bind="foreach: settings.settings.plugins.klipper.macros">
            <!-- ko if: tab -->
            <button class="btn btn-block btn-small" data-bind="text: name, click: $parent.executeMacro, enable: $parent.isActive()">E-Stop</button>
            <!-- /ko -->
         </div>
     </div>
    </div>
  </div>
</div>
                                </div>
```

</details> 

Thanks for your work on this plugin. Appreciated.

jneilliii, 2021-03-05:
Thanks for including the html of the tab. That will help isolate this more quickly. But prior to me putting much effort into this could you test out my curret development version that utilizes a different draggable library.  The process is a little different than before and you add the tabs from the main UI once installed. Copy/Paste the URL below into Plugin Manager > Get More > ...from URL and click Install. Once installed and OctoPrint restarts click the pencil icon in the navbar to enable edit mode on the consolidated tab.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

Sineos, 2021-03-06:
Wow, the gridstack stuff is really cool and .... it solves the copy / paste issue 👍 

Thanks a lot

closed 2021-03-06 by Sineos