jneilliii/OctoPrint-ConsolidatedTabs #47: [BUG]: Edit button disappearing after initial setup

R1N4x, 2021-03-18:
First of: I love the Plug-In.

But there is a weird bug. 
After I've set up the first layout and saved it, the edit button disappeared. (I've checked the settings!) 

I found the reason why: the button has the CSS style "Display: none" (See screenshot)

[image]

Running OctoPrint 1.5.3 and Python 3.7.3

jneilliii, 2021-03-18:
can you test with the latest rc version. I think this is a conflict with another plugin that I've already fixed.

jneilliii, 2021-03-18:
actually, sorry that was a different issue that I fixed. Could you check the developer console of your browser though and see if there are any errors on initial page load please.

R1N4x, 2021-03-19:
Sure, here you go: 

[image]

(For testing I disabled all Plug-Ins but yours)

R1N4x, 2021-03-19:
> can you test with the latest rc version. I think this is a conflict with another plugin that I've already fixed.

Installing 1.0.1rc1 fixed it.

closed 2021-03-19 by R1N4x

jneilliii, 2021-03-19:
Ok cool. It was the error that I fixed for people that don't have webcams.