jneilliii/OctoPrint-ConsolidatedTabs #49: [BUG]: ConsolidatedTabs is disabling PrettyGcode plugin Dashboard/State/File Windows in Full Screen mode

LarryVand, 2021-08-02:
**Describe the bug**
With ConsolidatedTabs (CT) and PrettyGCode (PGC) plugins installed, toggle PGC to full screen and try clicking the Dashboard/State/File buttons. Nothing happens. But disable ConsolidatedTabs plugin and the Dashboard/State/File buttons work just fine. 

**Expected behavior**

**Screenshots**

**Browser errors**

jneilliii, 2021-08-02:
Something must have changed since #38 was resolved. I assume you cross posted an issue on their plugin as well?

LarryVand, 2021-08-02:
Not yet, but I'm about to.

jneilliii, 2021-08-02:
Hey @LarryVand is the Pretty GCode tab consolidated or not?

LarryVand, 2021-08-02:
No it is not. It is in its own tab.

jneilliii, 2021-08-22:
I've just release version 1.0.2rc1 for you to test with. Change the plugin's release channel to `Release Candidate` in OctoPrint's Software Update settings and click update when prompted.

LarryVand, 2021-08-22:
> I've just release version 1.0.2rc1 for you to test with. Change the plugin's release channel to `Release Candidate` in OctoPrint's Software Update settings and click update when prompted.

Yup, it's fixed. Many thanks for all your work.

LarryVand, 2021-08-22:
The issue is closed.

closed 2021-08-22 by LarryVand