jneilliii/OctoPrint-ConsolidatedTabs #51: [BUG]:

dietlow, 2021-12-29:
I can't believe I spent two days on the shit and it still doesn't work. Tabs just show the default camera.

jneilliii, 2021-12-29:
Sorry, I don't respond to that form of communication. If you want help try to be nicer for something that I have provided for free and apparently works fine for a known 1.34k installs.

dietlow, 2021-12-29:
My apologies I was really frustrated last night. I got both streams working but when I plug them in to Multicam it just shows the default camera in both tabs.  I’m out of ideas

jneilliii, 2021-12-29:
So it's a conflicting issue with multicam plugin. Which one of those are you using, because I think there are a couple of them.

jneilliii, 2021-12-29:
and when you say both tabs, what do you mean by that?

dietlow, 2021-12-29:
MultiCam (0.2.8)

dietlow, 2021-12-29:
When I add  another camera to MultiCam with the working url that I tested in a browser window. All that happens is I see the default camera in both links/button.

jneilliii, 2021-12-29:
duplicate of https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/issues/20, I submitted a PR to that plugin to fix it...

jneilliii, 2021-12-29:
It appears my PR was merged, but the releases weren't updated. Try installing by copying/pasting this URL in Plugin Manager > Get More > ...from URL

```
https://github.com/mikedmor/OctoPrint_MultiCam/archive/refs/heads/master.zip
```

dietlow, 2021-12-29:
Thants a no go 
Still same symptoms.

jneilliii, 2021-12-30:
weird, just tested here and I'm able to switch the camera view in the consolidated tab. is your control tab not consolidated or is? if you open your browser's developer tools (F12) and look on the console tab are there any errors on load or when trying to switch cameras?

jneilliii, 2021-12-30:
https://autode.sk/3mHs3FP

dietlow, 2021-12-30:
Yes here is what I see

[Error] Stream Error, but src is empty.
	(anonymous function) (packed_plugins.js:8647)
	dispatch (packed_libs.js:13:43096)
[Warning] The resource http://192.168.2.185/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.2.185/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.2.185/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.2.185/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.2.185/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.2.185/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://192.168.2.185/static/vendor/font-awesome-3.2.1/fonts/fontawesome-webfont.woff?v=3.2.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (getSnapshot, line 0)
[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (getSnapshot, line 0)

dietlow, 2021-12-30:
Sorry no errors when I switch cameras the only thing I see here is octolap that is causing the error but that can't be the issue because I just installed it[

dietlow, 2021-12-30:
Attachment available until Jan 28, 2022

Click to Download
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAdFrP9zLKJv1p0BMGKVyDvs6qW8yAQBtZqdbLgbFVgYD3SpG1hoW5I2u%2F%24%7Bf%7D%3Fo%3DAh5ZR5L0Ka0r-YGi8MjiXowU_LJYDCOuH-j1tksmWwX0%26v%3D1%26x%3D3%26a%3DCAogATqXEPtQ9Y2RNefHK0veAVDo-r9RRSOq-QZp9JhKJPkSdhCd7JPK4C8YnfyOnuovIgEAKgkC6AMA_0mYLkNSBDqpbzJaBBbkja5qJaF0X5PRxHUEk6HZka6m-3-fY320gNnVEZwsvkwI2hFn4RqfjshyJSKK27aJczQVQTZlFQuuKo52FJviqFCDFr5tcEc-t1gCfHZ8o_w%26e%3D1643425021%26fl%3D%26r%3DDEE6A717-E94A-4142-BA3D-EAF49644993F-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3DA1CD9247-5F50-49EF-8D7C-7264DFD469CE%26p%3D27%26s%3DzLmwNghABK9Mxrmng-7z7H2zDBM&uk=qyJVUkEbwRF_xuA__-j0wA&f=screen%20.mov&sz=102988099>screen .mov
103 MB

dietlow, 2021-12-30:
Also got this on chrome

Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?0d093a46:1273 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?0d093a46:1273 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_update since it does not exist
packed_core.js?0d093a46:1273 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_settings since it does not exist
packed_core.js?0d093a46:1273 Did not bind view model bedlevelvisualizerViewModel to target #wizard_plugin_bedlevelvisualizer since it does not exist

jneilliii, 2021-12-30:
Yeah none of those errors seems relevant to Consolidated Tabs plugin, and the movie you shared seems to not have the plugin enabled?

dietlow, 2021-12-31:
What else do I have to do to enable the plug-in?

jneilliii, 2021-12-31:
Once this plugin is installed, go to it's tab (usually named OctoPrint or whatever you changed the Appearance > Title to in OP's settings), click the edit button in the navbar at the top, add the tabs to it, click the save button, reload. The readme doesn't show well that the first image is a link to YouTube...

https://youtu.be/1Qg6TIGGRB4

jneilliii, 2021-12-31:
Actually, the image in settings shows that process...

https://youtu.be/BuJJV-giFNc

dietlow, 2021-12-31:
Hi I think we're talking about the wrong plug-in I was talking about the multicam it's still not working

jneilliii, 2021-12-31:
Yeah, so that's not my plugin, so can't really help you there. Maybe you should post on that plugin's repository for assistance with it.

dietlow, 2021-12-31:
Are you kidding me within chatting back-and-forth about the wrong plug-in?
Lol 
Thanks for being so patient

closed 2021-12-31 by jneilliii