jneilliii/OctoPrint-ConsolidatedTabs #57: [BUG]: Pencil/Edit Icon Not Appearing on Toolbar

roomonthethird, 2023-11-21:
**Describe the bug**
Hello!

I downloaded Consolidated Tabs and do not see the edit button.

I see someone else had a similar issue in Issue #50 but I did not see any steps how to fix the issue.

I checked on and off different settings within the plugin and did not notice any change to the issue.

Things I tried to solve the issue:
- I ran Octoprint in [Safe Mode](http://docs.octoprint.org/en/master/features/safemode.html) and this did not solve the issue.
- I turned off all plug ins by process of elimination to see if that would affect anything. No change.
- I cleaned up the data (eraser button) of all third party plug ins. No change.
- Tried running Octoprint in Firefox (primary use browser), Chrome and Safari. All instances experienced the same issue.

**Expected behavior**
There should be a pencil icon in the navbar that allows me to edit the consolidated tabs plugin.

**Screenshots**
[image]

**Browser errors**
[image]
[image]

**Extra Info**
    OctoPrint version 1.9.3
    OctoPi* 0.18.0 (build 2022.01.20.12925)
    Python 3.7.3
    Heavily modded Ender 3 with a BIGTREETECH SKR Mini E3 V1.2 board
    Marlin 2.32 from TH3D
    Firefox, Chrome, Safari
    Windows 10 and macOS Ventura 13.6.2

jneilliii, 2023-11-21:
The error you show is pretty gcode based. The edit button only shows when the consolidated tab is active.

roomonthethird, 2023-11-21:
🤦 
Thank you so much, sorry for the dumb post!

closed 2023-11-21 by roomonthethird